bmorcelli/Launcher #130: CYD-3248S035R testing .

CoolmdXi, 2025-03-09:
Hi 
I used your WIP to the above board but there appears to be a touch issue, almost anywhere touched on screen seems to move forwards, cant find a position for previous and select is very random, so could get very far.
 Also my colors were incorrect the main display appears whiteish and when, somehow on one occasion I did get the select to work i selected blue and the overall effect was green.

I believe my board has ILI9488 driver and works ok with bb-spi_lcd and tft_espi

Many thanks and great work, i did try the JC4827W550 and got similar errors to Shadowtrance.

bmorcelli, 2025-03-10:
Hello!! Ok, the colours are an easy fix..

The display mapping is like this:
```
                    Touch area Map
            ________________________________ 0
            |   Esc   |   UP    |         |
            |_________|_________|         |_> third_y
            |         |   Sel   |         |
            |         |_________|  Next   |_> third_y*2
            |  Prev   |  Down   |         |
            |_________|_________|_________|_> third_y*3
            |__Prev___|___Sel___|__Next___| 20 pixel touch area where the touchFooter is drawn
            0         L third_x |         |
                                Lthird_x*2|
                                          
```

So not only the footer.. maybe it can help you telling me the orientation issue..

I will make the changes and enable the Serial Logs to determine the orientation of the touchscreen.. I'll let you know whenever I get it ready to test

bmorcelli, 2025-03-10:
available for testing.

please sen the serial logs, and tell me where you clicked for every place shown on the log

CoolmdXi, 2025-03-10:
The colors are now correct, but rotation and touch are not corresponding. poss 90 degrees wrong, hard to determine. the logs are not always appearing , i was trying to get the touch pressed so i could give you more but sometimes long after touch they appear. will try again laters. Thanks

bmorcelli, 2025-03-10:
I got some help on Discord for this device, I believe it might be working fine now..

can you test it?

bmorcelli, 2025-03-10:
Confirmed fixed ;)

closed 2025-03-10 by bmorcelli

CoolmdXi, 2025-03-12:
reply, i think there may be different versions of this board, because colors are still inverted and touch is not correct at all with my board. i think maybe there were different screen drivers for these boards but i thought the touch stuff on the resistive was the same.
my board has ILI9488 driver . and these are the pins for xpt2046
// Touchscreen pins
#define XPT2046_IRQ 36   // T_IRQ
#define XPT2046_MOSI 13  // T_DIN
#define XPT2046_MISO 12  // T_OUT
#define XPT2046_CLK 14   // T_CLK
#define XPT2046_CS 33    // T_CS