bmorcelli/Launcher #139: 87bb350 beta WebUI is faulty on T-Embed CC1101 / T-Dongle S3

mileshkin, 2025-03-19:
WebUI just shows this:

[image]

No buttons are clicklable and no actions are happening on any button strike.

bmorcelli, 2025-03-19:
Your browser isn't allowing JavaScript 

I just tested here and it is working on all my devices

mileshkin, 2025-03-19:
Hmmm... Strange thing. 
JS enabled... just tested Chrome / Android Chrome - the same nothing. 
.... while Bruce and 2.4.6 release / some of older betas of Launcher is ok...

bmorcelli, 2025-03-21:
indeed it was broken, it wasn´t loading all the javascript file.. i have just fixed it and improved the drag/drop and added support to folder upload

closed 2025-03-21 by bmorcelli