bmorcelli/Launcher #141: CYD-2432S028 (Resistive) touch is inverted

Amking123, 2025-03-20:
Hi and sorry about asking a lot of questions but can you help me to fix the inverted touch the CYD Is DIY it's with ESP32 wroom32 and TFT 2,8 with ILI934 .
Please

bmorcelli, 2025-03-21:
It is inverted how? In the width or in the height or both are inverted?

Amking123, 2025-03-21:
In the both like if you want to press next you should touch the up side of the screen

bmorcelli, 2025-03-21:
You will need to build the project locally, and add  this line:
```
        -D CYD28_TouchR_ROT=3
```

Here:
 https://github.com/bmorcelli/M5Stick-Launcher/blob/0c100f3c4be2943a20a5d2849bff44c661c056f2/boards/CYD-2432S028/platformio.ini#L22

And build the CYD-2432S028 environment 

` pio run -e CYD-2432S028`

closed 2025-03-22 by bmorcelli