bmorcelli/Launcher #143: WebUI File Uploading isnt working for M5StickCPlus2

tommyfart, 2025-03-22:
this is my second time trying to use M5Launcher but when i try uploading files using WebUI it keeps saying "Upload Complete!" but it didnt upload it

bmorcelli, 2025-03-22:
Which version have you installed?

bmorcelli, 2025-03-22:
I published 2.4.7 yesterday here on github with some improvements on the WebUi and I have just upload it to m5Burner..

tommyfart, 2025-03-22:
> I published 2.4.7 yesterday here on github with some improvements on the WebUi and I have just upload it to m5Burner..

i have 2.4.7 and i realized i need an SD Card, can you try making a WebUI version that uses LittleFS to store .bin files?

bmorcelli, 2025-03-22:
Not possible.. LittleFS/Spiffs is to be used by the installed firmware.. it is too small to save binaries, unfortunately

closed 2025-03-27 by bmorcelli