bmorcelli/Launcher #144: CYD-3248S035C (3.5in, Capacitive) - Inverted Colors

Glenn762702, 2025-03-26:
Hello, I have a CYD-3248S035C (3.5in, Capacitive). When installing the capacitive firmware, my colors are inverted and I notice pressing on the screen doesn't press where it's supposed to, almost like it's mirrored. I tried installing the Resistive just to see what would happen and the colors are normal, but it's not functional since I can't use the touch screen.

erwantestI23, 2025-03-26:
hello @bmorcelli 
tft.swapBytes(true); 
is defined in setup in the code ?
for color
Bye

bmorcelli, 2025-03-27:
>  my colors are inverted

I'm looking at it ryt now

>  I notice pressing on the screen doesn't press where it's supposed to, almost like it's mirrored. 

I've got a board here with this same simptom.. i'm looking at it too

>tft.swapBytes(true);
> is defined in setup in the code ?
> for color

@erwantestI23 these boards are using ArduinoGFX, it is a IPS true or false issue.. might be an easy fix

bmorcelli, 2025-03-30:
Can you test the BetaRelease version please?

Glenn762702, 2025-03-31:
> Can you test the BetaRelease version please?

The Beta release works flawlessly!

closed 2025-03-31 by Glenn762702