bmorcelli/Launcher #147: resting for the assistance  install it on an ELECROW ESP32

satishsezu, 2025-04-06:
Hello Team  can you help me install it on an ELECROW ESP32  2.4 inches, and is known as CROWPANEL WZ2432R024.ELECROW ESP32 display-2.4

[image] 

thank you in advance

bmorcelli, 2025-04-06:
Hello!!

I made a version for this board (and expanded to 2.8in and 3.5in), but as I don't have it to test, I need you help

If it doesn't work, I need you to send me the Serial Monitor Logs to analyse and find out how to fix it.

https://bmorcelli.github.io/M5Stick-Launcher/
[image]

bmorcelli, 2025-04-11:
can u test?

bmorcelli, 2025-05-10:
can you test it pleasE?