bmorcelli/Launcher #149: Touch misaligned CYD ESP32-2432s032C

wertsaco, 2025-04-08:
The touch screen is misaligned, the area being touched is not correct and does not correspond to the on-screen buttons.
I already tried the beta and stable version.

bmorcelli, 2025-04-08:
Hey friend..

Can you open a Serial Monitor and tell me the values you get when you press at the top left corner?

The serial message should be something like
```
Touch Pressed on x=%d, y=%d, rot=3
```

The desired values for x and y at top left corner are x=0 and y=0.

wertsaco, 2025-04-08:
hi, the corner just down the led is

Touch Pressed on x=313, y=7, rot=3
[ 75582][I][interface.cpp:219] InputHandler(): 
Touch Pressed on x=313, y=7, rot=3

the corner placed left at the usb C port is 

Touch Pressed on x=317, y=316, rot=3
[198018][I][interface.cpp:219] InputHandler(): 
Touch Pressed on x=317, y=316, rot=3

the corner just below the photoresist is 

Touch Pressed on x=84, y=6, rot=3
[409627][I][interface.cpp:219] InputHandler(): 
Touch Pressed on x=84, y=6, rot=3

and the corner at the right side of the USB C is 

Touch Pressed on x=84, y=316, rot=3
[460384][I][interface.cpp:219] InputHandler(): 
Touch Pressed on x=84, y=316, rot=3

hope this was useful, thnx

kbnhvn, 2025-04-26:
Hi, same board and same issue, any fix ?

bmorcelli, 2025-05-10:
Can you test the BetaRelease, please?

wertsaco, 2025-05-11:
awesome its working thnx

closed 2025-05-11 by wertsaco