bmorcelli/Launcher #153: Lilygo T5 S3 E-Paper Pro 4.7" Launcher Flash Causes Unit to Become Unresponsive

Wio-Steelhor, 2025-04-22:
Hello BMorcelli,

I received my Lilygo T5 S3 E-Paper Pro 4.7" unit recently and stumbled upon your Launcher project as my first usable firmware to flash to it.  The unit works normally with the factory sample firmware.  The Flasher for the T5_E-Paper_S3_Pro through the Launcher website appears to function normally using either the Latest or Beta Releases.  However, once it is complete, the unit is unresponsive and only shows the last image that was on the screen before the flash completed.  No single or combination of any of the four case buttons gives any response.  I can reflash the "H752_01_factory_v1.6_20250308_915M.bin" file and restore the factory sample, which responds normally.  But any attempt to reflash the Launcher firmware results in the same unpowered, unresponsive result.  Below is the little bit of information the factory image provides.  What else may I gather to help you troubleshoot this issue?

Results from "Settings, About System" from factory v1.6 image:
SF Version: v1.6-250308
HD Version: v.1.0-241224

bmorcelli, 2025-04-22:
Hey my friend!!

Launcher works with the older hardware version of the T5S3 e-paper pro only, at least for now, unfortunately..

Lilygo changed almost all the hardware from H752 to H752-01, that is the one you have..

Maybe I can make it work, even without the device, so I'll let you know when I have a version on the Beta>WIP section for testing.

jesshampshire, 2025-09-29:
Do you need one, to get the port going?

pelgraine, 2026-03-12:
Any progress with this one? I have the H752B (without GPS) and would love to be able to run Launcher on it.

pelgraine, 2026-04-01:
Sure is! Thanks for all your hard work! Extremely useful to have the Launcher on this device. Much appreciated.