bmorcelli/Launcher #169: how to get the sd card working for M5StickC Plus 2 + Sharkbait adapter

x0SiN0x, 2025-06-11:
Flashed the launcher on my M5StickC Plus 2 and the SD card is not recognized.  loaded the bruce firmware to confirm it works there being this sharkbait board was designed for that firmware in mind and the SD card shows up fine.  I went back to the launcher ui and attempted to set the pins to what i found here in the bruce code but still no sd card recognized.

any thoughts what I could be missing or maybe the sd pin settings not getting set?

https://github.com/pr3y/Bruce/blob/main/boards/m5stack-cplus2/pins_arduino.h
Tried these pins, wasnt sure what to put for CS so tried a few (-1, 0, 5, and 14)
```
MOSI = 15;
MISO = 36;
SCK = 13;
```

Also checked here https://github.com/pr3y/Bruce/blob/main/boards/m5stack-cplus2/m5stack-cplus2.ini and set according to the sd card setup still no luck
```
;SD Card Setup pins
-DSDCARD_CS=14
-DSDCARD_SCK=0
-DSDCARD_MISO=25
-DSDCARD_MOSI=26
```

bmorcelli, 2025-06-11:
What is this sharkbite adapter?

If it is using the shared SPI schematics, you would need to set the gpio 33 to output HIGH in the interface _setup_gpio()

bmorcelli, 2025-06-11:
Sure.. I'll add it..

bmorcelli, 2025-07-10:
Added on beta, fixed