bmorcelli/Launcher #173: Add support for T-LoRa pager

Szetya, 2025-06-19:
Hi! 

Would it be possible to add the Lilygo T-LoRa Pager in the future? 
It is still very new HW, there are not many applications available. Because of the peripherals (LoRa, NFC, GPS, etc) and the expansion port, I see a lot of potential to develop it in time.

https://lilygo.cc/products/t-lora-pager

What can complicate things is that the peripherals are connected to an IO expander, including the SD card.
https://github.com/Xinyuan-LilyGO/LilyGoLib/blob/master/docs/hardware/lilygo-t-lora-pager.md

bmorcelli, 2025-06-19:
Hey friend!

It will.. I have a unit coming, so I'll be able to mess with it and make it available in Launcher

Szetya, 2025-06-19:
You are fantastic and thank you very much!
I managed to get a SX1262 version, although I was hoping for a LR1121 version. 😂 
I haven't made much progress with it so far. I tried some ready binaries. The integration into meshtastich is very complicated for me with this IO expander thing. 🙁

PR #177 (2025-07-08): Add LILYGO T-LoRa-Pager port
Port for LILYGO T-LoRa-Pager.

~Not yet ported Bruce so I cannot test flashing.~

I went with the back/escape button being mapped to the middle button on the bottom, I did not think using the backspace button was a workable solution.

It uses the new version of XPowersLib which supports the BQ25896 power management chip.
Also ended up going with the TCA8418 Adafruit library to get the keyboard working.
The XL9555 GPIO expander uses the SensorLib.

Since the keyboard lacks some of the symbols I had to enable encoder and keyboard support in `mykeyboard.cpp`.

Quick v [... cut]
closed 2025-07-08 by bmorcelli