bmorcelli/Launcher #180: LilyGo T-Deck Pro Firmware

leebobleebob, 2025-07-09:
I have tried Both the Beta and the most recent firmware and it seems that I can run Launcher exactly 1 time and then no matter the firmware I load from the SD card it seems, I am unable to get back to choose anything on the SD card, no matter how many times I try. I am pressing the enter key and it doesn't seem to matter, this model does not seem have trackball like the other ones this is the one with the epaper display.

It is highly possible I am not pressing the correct buttons to get in, I have explored a number of them with no Joy any help on this would be greatly appreciated :) I love using this tool on my carputer, so thanks so much for all of the hard work!

leebobleebob, 2025-07-09:
Sorry just to be clear I was using the web flasher

leebobleebob, 2025-07-12:
OMG Thank you!

konomikitten, 2025-08-15:
I've been testing the keyboard patch on my device thanks for patch. Is there any chance we could get the following:

* Enter to interrupt the launcher at startup
* WASD navigation

Thanks in advance, if you need anyone to help testing I'd be happy to.