bmorcelli/Launcher #184: LilyGo T-Deck Pro USB mount not working on mac

leebobleebob, 2025-07-12:
Doesn’t seem to launch in Mac (sd card on its own mounts fine) not sure if this is intended to work.

It is visible in disk utility but not mounted and Even manual mode does not mount.

Also cannot get T-Deck pro to exit USB mode without reset of device. Tried tapping various screen locations and many buttons.

Thank you all so much for you very hard work on this amazing project :)

google906, 2026-01-18:
hello, similar or same problem here in my linux and latest m5launcher on my t-deck pus (not the pro) i built on vscode and platformio. works fine, but usb mode wont work as the device wont show up on my laptop. 

(i'm not good in guessing, but i jave to mention i added this platformio udev rules,.... hopefully they dont block the device from beeing recognized as storage (which i assume this function does)

thanks for your work.

benallfree, 2026-09-30:
@bmorcelli this one is still happening for me on the tdeck plus, have you heard any new info? running latest beta

bmorcelli, 2026-09-30:
@benallfree hey brother, I haven't heard about it..

It was mentioned on other issue, I tried to patch ot, but the guy never answered, something about it not being automatically mounted on Mac..

I don't have any Mac device to test and debug, so I meed someone to fix it for me

benallfree, 2026-09-30:
@bmorcelli I have a mac and hit a dead end yesterday. `diskutil` shows the device but Finder does not display it. The same SD card works from an SD card reader so I suspect something in the handshake. I will try again soon and report back.

konsumer, 2026-09-30:
@benallfree Make sure you are using latest firmware. I made [the PR](https://github.com/bmorcelli/Launcher/pull/297) that fixes this issue on mac when using other devices, and it should work here. I needed to turn off power on device before plugging in (which triggers specific "I am USB device" behavior.) I don't have a T-Deck Pro, so I can't test, but it definitely should work unless they do USB different.

> It was mentioned on other issue, I tried to patch ot, but the guy never answered, something about it not being automatically mounted on Mac..

@bmorcelli by "the guy" you mean me? I have no T-Deck Pro to test, but it fixed it on mac on several other devices here (regular tdeck, catrdputeradv, etc.) The PR that fixes this should be merged in current. I can test more with these devices, if it will help, but no T-Deck Pro here.

benallfree, 2026-10-01:
@konsumer I have a TDeck Plus and was unable to get it working from anything in this repo. It doesn’t seem to respond at all when powered off/on but I’ll try again because it sure would be nice to get to the bottom of it.

konsumer, 2026-10-01:
@benallfree how exactly did you install? Here is what I would do:

- go to [webflasher](https://bmorcelli.github.io/Launcher/webflasher.html) and choose Beta/LilyGo/T-Deck Pro ("ePaper display")
- unplug USB
- turn off power switch - maybe this is not needed for yours, but it was for my devices. They power on with USB, but it tells ESP32 "act like a USB device"
- plug in USB

Without having the device, I can't troubleshoot much more than this, but that should be the process to get it working. If it's still not, share these logs:

- `/Launcher_error.log` on SD
- mac logs

```
# A. Does the device exist on the bus at all?
system_profiler SPUSBDataType | grep -A15 -i "Launcher\|303a"
ioreg -p IOUSB -l -w 0 | grep -i -A30 launcher

# B. Live kernel log while plugging in (device present but not mounting)
log stream --predicate 'eventMessage CONTAINS "USBMSC"' --style compact
sudo log stream --predicate 'subsystem == "com.apple.iokit.IOUSBMassStorageDriver"' --style compact

# C. After the attempt (retrospective)
log show --last 10m --predicate 'eventMessage CONTAINS "USBMSC"' --style compact
sudo log show --last 10m --predicate 'process == "diskarbitrationd"' --style compact

# D. Is it a disk, and does bulk I/O actually work?
diskutil list
diskutil verifyVolume /dev/diskN     # issue #424: this passed while mount still failed
diskutil activity                    # live disk-event stream
```

konsumer, 2026-10-01:
I made a [tester](https://github.com/konsumer/msc-poc/) that just implements the USB/SD part. I don't have time right now to test on my own hardware, but try it out. You can get ready-made firmware images in [releases](https://github.com/konsumer/msc-poc/releases).

benallfree, 2026-10-01:
@konsumer Did you say there is firmware that does it already? That might be interesting, that would basically be like the "USB" feature of Launcher but installed separately?

Another topic, somewhat related - I'd love to find a way to get Launcher in my local development workflow, like flashing updated PlatformIO builds to an existing app partition. That is my actual use case that led to noticing the USB drive wasn't mounting properly.

konsumer, 2026-10-01:
Yeh, there is probly something existing, but I also made one that works with my cardputerADV (and should work with your T-Deck Pro) on mac, and only does that. This is the SD, in my carduter, which is pretending to be a USB drive:

[image]

Again, this just uses the same code as Launcher, so if it works, Launcher should work too.

> Another topic, somewhat related - I'd love to find a way to get Launcher in my local development workflow, like flashing updated PlatformIO builds to an existing app partition. That is my actual use case that led to noticing the USB drive wasn't mounting properly.

Yes, I originally wanted similar. Honestly, what I do now is pio normally, and just flash direct to try stuff out, then when it's all ready to go, I reflash Launcher. It's a bit annoying to swap them, but they sort of inhabit different usecase-spaces.

I worked out this whole complicated build/copy to SD setup in my `platformio.ini`, but there isn't a direct way to say "ok, now load and run this" and also you get into weird overlap when you need to swap between "pretend to be a USB thumbdrive" and "go into boot debug mode so I can upload code", not to mention whatever the firmware does (like I do radio stuff and emulate other USB things) and it gets messy. At first I was a bit annoyed to keep having to reflash Launcher when I was done developing, but it's not too bad (you can even script it, and upload fairly quickly with the bin, directly.)

I think what made me more comfortable with this workflow is just realizing all the tooling is made to work like this, and Launcher is not really for that (it's just a great way to swap out lots of finalized firmwares.) The time I spent fighting my tools would have been better spent writing code!

konsumer, 2026-10-01:
#439 seems to fix current code on cardputer/mac (SPI was too slow to work correctly) so that might also be related. I still don't recommend using USB to juggle firmwares, though (just use pio the normal way, and reflash Launcher when you're done.) It's way easier to automate (reset-to-debug, flash, reset just using normal pio stuff) and no messy stuff with USB or anything else.

bmorcelli, 2026-10-01:
@benallfree Available for testing on Beta

@leebobleebob

konsumer, 2026-10-01:
@benallfree the above PR helps with dev on firmwares. It makes it so you can push into Launcher, automatically (instead of replacing firmware) using a script. Basically put `tools/pio_launcher_upload.py` in your project and add this to platformio:

```yml
extra_scripts = post:./tools/pio_launcher_upload.py
``` 

Then when you do `pio run -t upload` it will automatically push it into Launcher, instead.