bmorcelli/Launcher #185: T.Display AMOLED PLUS error

fermintm, 2025-07-13:
HI. T-display amoled plus not work. 
I guess you've got the wrong model. The T-display AMOLED Plus is this one: https://lilygo.cc/products/t-display-s3-amoled-plus

bmorcelli, 2025-10-30:
Hello, can you try the build again? 

Beta Release > WIP > T-Display S3 AMOLED Plus

[image]

fermintm, 2025-10-30:
[21:39:48]ESP-ROM:esp32s3-20210327
[21:39:48]Build:Mar 27 2021
[21:39:48]rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
[21:39:48]Saved PC:0x4037a182
[21:39:48]SPIWP:0xee
[21:39:48]mode:DIO, clock div:1
[21:39:48]load:0x3fce2820,len:0x11e4
[21:39:48]load:0x403c8700,len:0xc74
[21:39:48]load:0x403cb700,len:0x3008
[21:39:48]entry 0x403c88c0
[21:39:48][boot] Turned on because (1= POWERON_RESET or 5==ESP_RST_DEEPSLEEP) (Other= Probably forced by launcher)--> 21
[21:39:48]E (25) esp_image: image at 0x200000 has invalid magic byte (nothing flashed here?)
[21:39:48]E (31) boot: OTA app partition slot 0 is not bootable
[21:39:49]SDCARD mounted successfully
[21:39:50]Devices Scan start.
[21:39:50]I2C device found at address 0x15 !
[21:39:50]I2C device found at address 0x51 !
[21:39:50]I2C device found at address 0x6B !
[21:39:50]Done
[21:39:50]
[21:39:50]Devices Scan start.
[21:39:50]I2C device found at address 0x15 !
[21:39:51]I2C device found at address 0x51 !
[21:39:51]I2C device found at address 0x6B !
[21:39:51]Done
[21:39:51]
[21:39:51]Press the button to enter the Launcher!
[21:39:51]Press the button to enter the Launcher!
[21:39:52]Press the button to enter the Launcher!
[21:39:52]Press the button to enter the Launcher!
[21:39:53]Press the button to enter the Launcher!
[21:39:53]Press the button to enter the Launcher!
[21:39:54]Press the button to enter the Launcher!
[21:39:54]Press the button to enter the Launcher!
[21:39:55]Press the button to enter the Launcher!
[21:39:55]Press the button to enter the Launcher!
[21:39:56]Press the button to enter the Launcher!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:39:57]Home key pressed!
[21:40:08]
[21:40:08]Pressed x=320 , y=68, rot: 3
[21:40:08]Pressed x=284 , y=68, rot: 3
[21:40:08]Pressed x=130 , y=79, rot: 3
[21:40:09]Pressed x=104 , y=124, rot: 3
[21:40:09]Pressed x=439 , y=163, rot: 3
[21:40:09]Pressed x=150 , y=101, rot: 3
[21:40:12]Pressed x=299 , y=177, rot: 3
[21:40:12]Pressed x=340 , y=72, rot: 3
[21:40:13]Pressed x=278 , y=92, rot: 3
[21:40:13]Pressed x=229 , y=132, rot: 3
[21:40:13]Pressed x=228 , y=177, rot: 3
[21:40:13]Pressed x=439 , y=183, rot: 3
[21:40:13]Pressed x=515 , y=175, rot: 3
[21:40:14]Pressed x=468 , y=86, rot: 3
[21:40:14]Pressed x=270 , y=36, rot: 3
[21:40:14]Pressed x=159 , y=92, rot: 3
[21:40:14]Pressed x=149 , y=169, rot: 3
[21:40:15]Pressed x=233 , y=221, rot: 3
[21:40:15]Pressed x=365 , y=219, rot: 3
[21:40:15]Pressed x=398 , y=116, rot: 3
[21:40:15]Pressed x=246 , y=66, rot: 3
[21:40:15]Pressed x=154 , y=145, rot: 3
[21:40:16]Pressed x=138 , y=190, rot: 3
[21:40:16]Pressed x=149 , y=207, rot: 3
[21:40:16]Pressed x=237 , y=88, rot: 3
[21:40:16]Pressed x=317 , y=50, rot: 3
[21:40:17]Pressed x=409 , y=100, rot: 3
[21:40:17]Pressed x=254 , y=208, rot: 3
[21:40:17]Pressed x=138 , y=115, rot: 3SDCARD mounted successfully

BLACK SCREEN :(