bmorcelli/Launcher #186: Porting to 4827S043r

fearx2024, 2025-07-16:
Hello, thanks to everyone working on the ESP32. I really appreciate your efforts and am a big fan. Since I also have a CYD, I would like to test the launcher, Bruce, and the Maruder firmware. Unfortunately, my CYD isn't supported. Can someone help me port it? I have a 4827S043r. Is there a problem porting it? The documentation is actually complete, enough to be able to evaluate it.[ ](https://github.com/rzeldent/platformio-espressif32-sunton)

bmorcelli, 2025-07-29:
it will be Available for testing in a few minutes

go to the Flasher and choose Beta Release > WIP > CYD-4827S043R

there are a big change for the touchscreen not be good, so we will need to work together to fix it.

bmorcelli, 2025-08-04:
Can you test it?

fearx2024, 2025-08-07:
Hello, my friend. So, I flashed the launcher. Unfortunately, the resolution is wrong and the screen is flickering. I tried that too and got the screen working halfway. But the touchscreen doesn't work, just like with your firmware. It has XPT2046 if that helps. I'll upload my source code too; I did it with Warp because I'm still learning.

bmorcelli, 2025-08-07:
It was using other board json information by mistake.. 

I´ve just fixed it now, should be working now.. and touchscreen must be working as well (might need to be calibrated)

you can test it on Beta

fearx2024, 2025-08-09:
Hello. So, I tested it. The display works. However, the touchscreen doesn't work at all. I'll send you the logs, maybe it will help. The touch controller is an xpt2046.
I have one more question: could you also port the Bruce firmware? I tried using Warp, but I couldn't integrate the parallel display.

 `E (103469) gpio: gpio_set_level(227): GPIO output gpio_num error
E (103475) gpio: gpio_set_level(227): GPIO output gpio_num e03744) gpio: gpio_set_level(227): GPIO output gpio_num error

E (108767) gpio: gpio_set_level(227): GPIO output gpio_num error
E (108773) gpio: gpio_set_level(227): GPIO output gpio_num error
E (108779) gpio: gpio_set_l�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x518
load:0x403c9700,len:0xc08
load:0x403cc700,len:0x2a0c
entry 0x403c98dc
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 1
E (41) ledc: ledc_set_duty(725): LEDC is not initialized
E (41) ledc: ledc_update_duty(653): LEDC is not initialized
E (962) gpio: gpio_set_level(227): GPIO output gpio_num error
E (2161) gpio: gpio_set_level(227): GPIO output gpio_num error

bmorcelli, 2025-09-13:
and did the beta work?
I believe that the touchscreen on 4827S043R should be working now

bmorcelli, 2025-09-14:
?