bmorcelli/Launcher #191: Oriantation problem for CYD-2432S024R

altunixcnc-oss, 2025-08-16:
Hi, first of all thank you for your great work on the Launcher.

I’m using a CYD-2432S024R and I’ve run into a problem:
when I download the firmware on to my esp the screen is set to landscape orientation, the problem is that touch input is still mapped as portrait. This makes the touches appear in the wrong place.

Could you please check if the touch mapping can be synced with the display rotation, or let me know if there’s a workaround? I’ve tried calibrating, but the issue persists.

Thanks a lot for your help and for maintaining this awesome project!

bmorcelli, 2025-08-19:
I will take a look on that, Thank you for reporting

Theres something to observe, if you are using GUITION CYD, try installing Custom Boards > RabbitLabs The Phantom

This board uses the GUITION 2432S024R (Same as Marauder), and the inputs are different from the regular CYD-2432S024R, 

I have both boards and they are different in the touch sensor orientation.

bmorcelli, 2025-09-25:
you probably have a GUITION board.. the colors are inverted in this board.. and the touchscreen is fully inverted

install the Custom Board > RabbitLabs The Phantom